heliosproj/HeliOS #10: Coop does now work on UNO

kwrtz, 2020-08-26:
I get follwowing result on UNO:

ACBACBACBL
ACBACBACBL
ACBACBACBL
ACBACBACBL
ACBACBACBL
ACBACBACBL
ACBACBACBL
ACBACBACBL

while using following code:

`/*
 * HeliOS Embedded Operating System
 * Copyright (C) 2020 Manny Peterson <me@mannypeterson.com>
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 */

/*
 * Additional documentation on HeliOS and its Application
 * Programming Interface (API) is available in the
 * HeliOS Programmer's Guide which can be found here:
 *
 * https://github.com/MannyPeterson/HeliOS/blob/master/extras/HeliOS_Programmers_Guide.md
 */

/*
 * Include the standard HeliOS header for Arduino sketches. This header
 * includes the required HeliOS header files automatically.
 */
#include "HeliOS_Arduino.h"

/*
 * The task definition for taskShort() which
 * will perform 100,000 arbitrary floating point
 * operations.
 */
void taskShortA(int id_) {
  float a = 0.0f;

  for (int32_t i = 0; i < 1000; i++)
    a *= 3.14f;

  Serial.print("A");
}

void taskShortB(int32_t id_) {
  float a = 0.0f;

  for (int32_t i = 0; i < 10000; i++)
    a *= 3.14f;

  Serial.print("B");
}

void taskShortC(int32_t id_) {
  float a = 0.0f;

  for (int32_t i = 0; i < 40000; i++)
    a *= 3.14f;

  Serial.print("C");
}

/*
 * The task definition for taskLong() which
 * will perform 1,000,000 arbitrary floating point
 * operations.
 */
void taskLong(int32_t id_) {
  float a = 0.0f;

  for (int32_t i = 0; i < 40000; i++)
    a *= 3.14f;

  Serial.println("L");
}

void setup() {
  /*
   * Declare and initialize an int to hold the
   * task id.
   */
  int id = 0;

  Serial.begin(9600);

  /*
   * Call xHeliOSSetup() to initialize HeliOS and
   * its data structures. xHeliOSSetup() must be
   * called before any other HeliOS function call.
   */
  xHeliOSSetup();

  /*
   * Add the task taskShort() to HeliOS by passing
   * xTaskAdd() the friendly name of the task as well
   * as a callback pointer to the task function.
   */
  id = xTaskAdd("TASKSHORTA", &taskShortA);

  /*
   * Call xTaskStart() to start taskShort() by passing
   * xTaskStart() the id of the task to start.
   */
       xTaskStart(id);

      id = xTaskAdd("TASKSHORTB", &taskShortB);
      xTaskStart(id);

      id = xTaskAdd("TASKSHORTA", &taskShortC);
      xTaskStart(id);

  /*
   * Add the task taskShort() to HeliOS by passing
   * xTaskAdd() the friendly name of the task as well
   * as a callback pointer to the task function.
   */
  id = xTaskAdd("TASKLONG", &taskLong);

  /*
   * Call xTaskStart() to start taskLong() by passing
   * xTaskStart() the id of the task to start.
   */
  xTaskStart(id);

    Serial.println("START");
}

void loop() {
  /*
   * Momentarily pass control to HeliOS by calling the
   * xHeliOSLoop() function call. xHeliOSLoop() should be
   * the only code inside of the sketch's loop() function.
   */
  xHeliOSLoop();
}`

ThomasHornschuh, 2020-08-26:
Your loop code will most likey optimized away bei GCC: Because a is never used, it does not need to be calculated. Even if you try to use a (e.g. printing it after the loop), the compiler will notice that it is essentially a constant expression (0*3.14) and will calculate this at the compile time and will essentially condense your loop to something like ```printf("%f\n",0)```
Do ensure that your loop is not optimized away you should
* declare the target variable as ```volatile```
* ensure the expression is depending on the loop variable (e.g. a*= i)

To be sure that your code is not optimized away use the gnu objdump utility ```objdump -d <elffile>``` and check the generated assembler code.

BTW: You tried to markup your code but to do it right you need three back ticks instead of one. Check with the preview function if your markup works as intended before posting. 
For longer code pieces you can also create a github gist (gist.github.com) and just store here

kwrtz, 2020-08-26:
Ok, thank you very much.  I didn't thought about the optimisation. Now it works :)

MannyPeterson, 2020-08-26:
@kwrtz  I will fix the Coop example Arduino sketch in this release. I don't want other's to get confused by the optimizations GCC is performing behind the scenes. I will fix the code as @ThomasHornschuh suggested. Thank you.

MannyPeterson, 2020-08-26:
Sorry for re-opening again. I will fix the example Arduino sketch as shown below. Thanks.

```
void taskShortA(int id_) {
  volatile float a = 0.0f, b = 0.0f;

  for (int32_t i = 0; i < 10000; i++)
    a += i;

  b = a;

  Serial.print("A");
}
```

closed 2020-08-26 by MannyPeterson