heliosproj/HeliOS #20: Dynamically Timed Tasks?

niektb, 2020-10-27:
Hi all!

I'm trying this RTOS to write the software for a digital guitar effect pedal but I ran across some issues (other than the docs not being up-to-date) with regards to a dynamically timed task...

First, I have a timed task that runs every 10ms which reads a pot value (between 0 and 1023) and calculates a delay tempo from it. This sets the timer of the next task accordingly.
```cpp
long delay_time_us = long(map(delay_time, 0, 1023, 53, 626)) * 1000;
xTaskSetTimer(xTaskGetId("TASKTEMPO"), delay_time_us);
```
This "TASKTEMPO" is a very simple task that toggles a LED:
```cpp
void taskTempo(xTaskId id_) {
  if (smode == SMODE_TEMPO)
    analogWrite(pin_rgb_g, 127 * tempo_blink);

  tempo_blink = !tempo_blink;
}
```

If I only have those two tasks, it works fine. However, when I add a couple more tasks, things start to get a little different. The timing seems still alright but often the task completely skips... Meaning that the LED blinks irregularly... Based on the concept of cooperative scheduling I would assume that the priority is high (as the taskTempo is very short) but it doesn't seem to be the case... Can I somehow assign priorities to tasks or is there a different solution to the issue?

MannyPeterson, 2020-10-29:
@niektb I can be more helpful if I can see your code. Are you able to post it? Also, what dev board/MCU are you targeting?

niektb, 2020-10-31:
Of course I can! Sorry if it's a bit rough round the edges but it is a heavy work-in-progress (currently working to remove a SPI digipot so you might still see some remnants in the code) :)

Right now I'm targeting an Arduino Nano (with the old bootloader)!

```cpp
#include <HeliOS_Arduino.h>
#include <avr/pgmspace.h>
#include <SPI.h>

/* PIN DEFINITIONS */
const uint8_t pin_footswitch  = 2;
const uint8_t pin_rgb_b       = 3;
const uint8_t pin_rgb_r       = 5;
const uint8_t pin_rgb_g       = 6;
const uint8_t pin_delay_mod   = 9;
const uint8_t pin_delay_pwm   = 10;
const uint8_t pin_tails       = 8;
const uint8_t pin_delay_time  = A0;
const uint8_t pin_mod_speed   = A1;

/* MODE PARAM */
bool bypass = true;
bool tempo_blink = false;

const uint8_t SMODE_BYPASS  = 0;
const bool BYPASS_NORM    = 0;
const bool BYPASS_TAIL    = 1;

const uint8_t SMODE_TEMPO   = 1;
const bool    TEMPO_POT     = 0;
const bool    TEMPO_TAP     = 1;

const uint8_t SMODE_MOD     = 2;
volatile double mod_speed     = 0.5;

const uint8_t MODE_CONFIG = 0;
const uint8_t MODE_PLAY   = 1;
volatile uint8_t mode = MODE_PLAY;
volatile uint8_t smode = SMODE_TEMPO;

/* MODULATION PARAMETERS */
const uint8_t TRIANGLE = 0;
const uint8_t SINE = 1;
const uint8_t SQUARE = 2;
//const uint8_t SAW = 3;
volatile uint8_t waveform = TRIANGLE;

const PROGMEM uint8_t triangle_wave[1024] = {/*ommitted for readability*/};
const PROGMEM uint8_t sine_wave[1024] = {/*ommitted for readability*/};
const PROGMEM uint8_t square_wave[1024] = {/*ommitted for readability*/};
volatile double wp = 0;

/* DELAY PARAMETERS */
#define MCP_NOP 0b00000000
#define MCP_WRITE 0b00010001
#define MCP_SHTDWN 0b00100001
const uint8_t ssMCPin = 10;

volatile int delay_time = 250;
volatile int delay_time_prev = 0;
unsigned long t_ms = 0;
unsigned long prev_t = 0;

unsigned long first_tap_time;
unsigned long first_release_time;
bool wait_for_release = false;

/* DEBOUNCE PARAMETERS */
const uint8_t debounce_delay = 20;
unsigned long last_debounce_time = 0;
[... 206 lines cut ...]
  xTaskWait(id);

  id = xTaskAdd("TASKMOD", &taskMod);
  xTaskWait(id);
  xTaskSetTimer(id, 4000); // 4ms

  id = xTaskAdd("TASKDELAY", &taskDelay);
  xTaskWait(id);
  xTaskSetTimer(id, 10000); // 10ms

  id = xTaskAdd("TASKTEMPO", &taskTempo);
  xTaskWait(id);
  delay_time = analogRead(pin_delay_time);
  long delay_time_us = long(map(delay_time, 0, 1023, 53, 626)) * 1000;
  xTaskSetTimer(id, delay_time_us);

  id = xTaskAdd("TASKSERIAL", &taskSerial);
  xTaskWait(id);
}

void loop() {
  // put your main code here, to run repeatedly:
  xHeliOSLoop();
}
```

niektb, 2020-11-11:
Any clues as to where this might go wrong?

MannyPeterson, 2020-11-12:
So one quick thing is to make sure you free the mem for tres:

xTaskGetInfoResult tres = xTaskGetInfo(xTaskGetId("TASKTEMPO"));

So taskTempo() starts to execute irregularly right? Have you tried adding one task at a time to see which task may be the culprit? It may be taskSerial() as serial IO takes a lot of clock ticks.

As far as priority goes. Tasks in a waiting state (event driven) always get priority in the scheduler. Tasks in a started state (cooperative) are not always scheduled for execution each cycle. But it appears you only have one cooperative task - taskButton().

Try adding tasks one by one to see if you can find the culprit and let me know what you find. My apologies for the delay getting back to you. I will keep my eyes open for your response.

MannyPeterson, 2020-11-12:
@niektb one other thing I noticed is you call xTaskSetTimer() in taskDelay(). Calling xTaskSetTimer() resets the timer. 

But it seems you have the below code block to prevent unintended resets.

if (abs(delay_time - delay_time_prev) < 3)
    return;

Make sure this isn’t causing unintended resets.

niektb, 2020-11-13:
@MannyPeterson No probs for the delay! Thanks for coming back to me! 

> So one quick thing is to make sure you free the mem for tres:
> xTaskGetInfoResult tres = xTaskGetInfo(xTaskGetId("TASKTEMPO"));

Hmmm, I added the serial.println for debugging in that if-statement. I indeed forgot to free the mem, the issue persists with or without :/ 

> So taskTempo() starts to execute irregularly right? Have you tried adding one task at a time to see which task may be the culprit? It may be taskSerial() as serial IO takes a lot of clock ticks.

Yes indeed! taskSerial isn't called atm (as I removed all debugging print statements via that task, to rule that out :)) it definitely seems to help to reduce the amount of tasks, but couldn't find a particular task that breaks everything :)

> As far as priority goes. Tasks in a waiting state (event driven) always get priority in the scheduler. Tasks in a started state (cooperative) are not always scheduled for execution each cycle. But [...]

But if multiple tasks are waiting, which tasks get priority? If timed tasks are configured to run at the same interval, which task gets the priority? The one that's started first?

> @niektb one other thing I noticed is you call xTaskSetTimer() in taskDelay(). Calling xTaskSetTimer() resets the timer.
> 
> But it seems you have the below code block to prevent unintended resets.
> 
> if (abs(delay_time - delay_time_prev) < 3)
> return;
> 
> Make sure this isn’t causing unintended resets.

Hmm good point, when I tested it, the ADC reading didn't fluctuate any more than '1', but maybe it's become more noisy since I made other changes in the circuit.

Is there a way to update the timer value without resetting it? That would probably fix the issue as well :)

MannyPeterson, 2022-01-20:
@niektb going to look at adding a function call to update the timer value. Adding to the development backlog.

MannyPeterson, 2022-01-23:
@niektb my apologies for the time it has taken to get to this - life distractions unfortunately have kept me away. In the next release there will be a function call that allows the user to sent the task timer interval without resetting the start time. You will see it in 0.2.8. Closing this for now. Thanks!

```C
void xTaskSetTimerWOReset(TaskId_t, Time_t);
```

niektb, 2022-02-19:
Cool! So what I'm curious about; what is the behaviour if you set the timer to a time that already is expired? :)

closed 2022-02-21 by MannyPeterson