heliosproj/HeliOS #24: Passing Parameters to task

niektb, 2021-06-19:
Hi! 
I had a question: is it possible to pass parameters to a task you're creating? 
So for example have two buttons but with a single task definition?
`void taskButton(xTaskId id_, const uint8_t port)`
which then is called with something like:
`
int p = 1;
id = xTaskAdd("TASKSW1", &taskSW, p)
`

MannyPeterson, 2022-01-20:
@niektb not currently, but I am going to add this to the development backlog as I like the idea.

MannyPeterson, 2022-01-24:
This feature will appear in 0.2.8

closed 2022-01-24 by MannyPeterson

MannyPeterson, 2022-02-15:
@niektb all done! Task parameters are now available in 0.3.0. Enjoy! :)

niektb, 2022-02-19:
Sorry for getting back so late! Great news! I now worked around it using classes but this way is much more compact and elegant :D