heliosproj/HeliOS #34: "Blink" example doesn't blink

d13g4, 2022-07-16:
Hi!
I loaded the example code in the arduino ide for blink and it didn't do anything.
I then loaded the basic example code for blink and it worked without problems.
I didn't change anything in the example - should i have?
I am using an arduino nano and i know HeliOS is about the limit for it, might that be a problem?
I have no experience with OS on microcontrollers and previously did everything "by hand".

MannyPeterson, 2022-07-17:
@d13g4   No changes to the code should be necessary for the Arduino Nano. I will build the Blink example on an ATMega328 based board using the Arduino IDE and get back to you. Thank you for bringing the issue to my attention. Manny

MannyPeterson, 2022-07-17:
@d13g4  just a thought, try changing the line

xTaskChangePeriod(blink, 1000000);

to

xTaskChangePeriod(blink, 1000);

See if that fixes it.

d13g4, 2022-07-17:
Yes!
Thats it, thank you.
I changed it to 10000 before, but obviously I wasn't patient enough.

MannyPeterson, 2022-07-17:
@d13g4 yeah that is a bug in the example. I will fix on next release so others don't have the same issue. Thank you.

sakunamary, 2022-07-22:
Hi Guys! 
   I have tried 3 kinds of MCU : amega328P (16Mzh), esp8266(80Mzh), and lgt8F328P(16Mzh,and 32Mzh) .
   The same example codes "blink" (I have already  fixed the bug "xTaskChangePeriod(blink, 1000);" ).

The result  as following :
   amega328P  OK ,of course 
  Lgt8F328P     OK ,MCU clock run on both 16Mhz and 32Mhz 
  esp8266EX (board is WEMOS D1 mini )       NOT working , MCU clock run on neither 80Mzh nor 160Mhz ... the led is flashing ,but too fast! maybe flashing every 500ms or 300ms . 
  
So ,for esp8266  ,should I confing some parms. or do someting setting jobs first ? 

Thank you and best wishes!

MannyPeterson, 2022-07-26:
I will look into the ESP8266 issue. Thank you for brining it to my attention! 😀

MannyPeterson, 2022-08-23:
@sakunamary the bug in HeliOS on the ESP8266 is now fixed. When I release 0.3.4 the fix will be included. The release should be available this week. Thanks!

closed 2022-08-23 by MannyPeterson