heliosproj/HeliOS #41: What is the difference between xTaskResume() and xTaskWait()  in function setup()?

sakunamary, 2023-02-24:
Hi:
    Here is my case:
    I am trying to make a canbus thermocouple module  with HeliOS ; the working flow as follow:
    step1: power up and run init scripts for init 
    step2: create two task 1)get temperature  2)write canbus data and send out 
         wait for 750ms and make sure MAX6675 get the correct  temperature data ,IN the meanwhile  canbus data will NOT send out 
    step3:     Task1:get temperature will run every 750ms and write the data in a float struct , notify task2 send out data
                    Task2:canbus data will send out data .

After I read thought  examples pgm :Blink.ino and coop.ino .I don't quiet to understand the difference between  xTaskResume() and xTaskWait() 
     In Blink.ino ,after create and check the task is OK . 
      xTaskWait(blink);    -------->Is that means task in suspended mode ,not running  
     xTaskChangePeriod(blink, 1000); ------>that means wait for 1s , as a timer ,like function delay()  of arduino ? 

     In coop.ino,after create and check the task is OK . 
       xTaskResume(shortTask); -------->Is that means task run immediately？

For further study,How to use MUTEX (in freeRTOS ) to lock the data?

Thank you for you guys answering!!!
Best wishes!!!

MannyPeterson, 2023-02-25:
@sakunamary 

The best place for information about HeliOS's system calls is the [HeliOS Developer's Guide](https://github.com/MannyPeterson/HeliOS/blob/master/doc/HeliOS_Developers_Guide.pdf).

xTaskResume() sets the task to "running". So the task WILL NOT respond to events. The task IS continuously running.

xTaskWait() sets the task to "waiting". So the task WILL respond to events. The task IS NOT continuously running. Instead, the task waits until either it receives a direct-to-task notification OR a task timer elapses.

xTaskSuspend() sets the task to "stopped". So the task WILL NOT respond to events AND the task IS NOT continuously running.

xTaskChangePeriod() sets the task timer to whatever period/interval you want. If the task is waiting (see xTaskWait()), when the timer elapses the task will run once and then wait again.

You do not need to use mutex's in HeliOS like FreeRTOS. You will not encounter concurrency issues when accessing data.

I hope this helps!

Manny

sakunamary, 2023-02-25:
Thanks for your answer!!  
Now I am clear!

closed 2023-02-25 by sakunamary