heliosproj/HeliOS #42: Can not get Notify messages from xTaskNotifyTake()

sakunamary, 2023-02-24:
Hi guys:
  I am trying to make a canbus thermocouple module with HeliOS ; the working flow as follow:
step1: power up and run init scripts for init
step2: create two task 1)get temperature 2)write canbus data and send out
wait for 750ms and make sure MAX6675 get the correct temperature data ,IN the meanwhile canbus data will NOT send out
step3: Task1:get temperature will run every 750ms and write the data in a float struct .And then notify task2 to  send data out 

I tested  example-- WaitNotify.ino  ,it work well . so I write my pgm base on WaitNotify.ino,here is source files

[sourcefile.zip](https://github.com/MannyPeterson/HeliOS/files/10821890/sourcefile.zip)

The main issue is   xTaskNotifyTake(task_) can  not attached ,not working 

Task1  send out notifications :

void taskThermo_Get(xTask task_, xTaskParm parm_) 
    {
  /* Get the task handle for the task that will receive the
  direct-to-task notification. */
  xTask WriteCANBUSTask = xTaskGetHandleByName("WRITECANBUS");

        temperature_data.BT_AvgTemp=30.12;
        temperature_data.ET_CurTemp=28.66; 

        Serial.print("task thermo_get");
        Serial.print("  ") ;
        Serial.print(temperature_data.BT_AvgTemp);
        Serial.print("  ") ;
        Serial.print(temperature_data.ET_CurTemp);
        Serial.println(" ");

        if (WriteCANBUSTask){  
            //Serial.println("geted task_writecanbus handler"); 
            /* Send the direct-to-task notification of "HELLO"
            which is five bytes. */
            Serial.println("task WriteCANBUSTask send noti "); --------> it's work 
            xTaskNotifyGive(WriteCANBUSTask,2, "OK");
        }
    }

 task2 :Receiver notifications 
void taskCanbus_Write(xTask task_, xTaskParm parm_) 
    {
  String notif_str = "";   
  Serial.print("before notif_str");
  Serial.println(notif_str); 
  xTaskNotification notif = xTaskNotifyTake(task_); ------>here ,not working .
notif_str = notif->notificationValue;------>here ,not working .
  Serial.print("after notif_str");
  Serial.println(notif_str);------->print out nothing...which is suppose is 'OK'

if ( notif ){. -------> here is not working neither 
    notif_str = notif->notificationValue;-------> here is not working neither 
        Serial.println(notif_str);-------> here is not working neither 
        xMemFree(notif);-------> here is not working neither 
}
    
The rest of codes is the same as example ino file  but the task name....
create task as below:
  xTask GetThermoTask = xTaskCreate("GET_THERMO", taskThermo_Get,NULL);
  xTask WriteCANBUSTask = xTaskCreate("WRITECANBUS", taskCanbus_Write, NULL);

and then  set task in waiting mode :
    xTaskWait(GetThermoTask);
    xTaskWait(WriteCANBUSTask);
and  set sending notify task in running mode which  loop  in 1s.
    xTaskChangePeriod(GetThermoTask, 1000); //1000ms 

Thank you for you guys answering!!!
Best wishes!!!

MannyPeterson, 2023-02-25:
@sakunamary 

Let's start with the following code for taskCanbus_Write().

```C
void taskCanbus_Write(xTask task_, xTaskParm parm_) {
  int bytes = 0;
  xTaskNotification notif = xTaskNotifyTake(task_);
  bytes = notif->notificationBytes;
  Serial.println("Notification Bytes: " + bytes);
  xMemFree(notif);
}
```

The output should be as follows.

```
Notification Bytes: 2
```

Please let me know if your output matches mine.

Thank You,

Manny

sakunamary, 2023-02-25:
Hi

> @sakunamary
>
> Let's start with the following code for taskCanbus_Write().
>
> ```c
> void taskCanbus_Write(xTask task_, xTaskParm parm_) {
>   int bytes = 0;
>   xTaskNotification notif = [...]

Hi Manny 
My output match your result 
Notification Bytes: 2
seem xTaskNotifyGive send out a message

sakunamary, 2023-02-25:
Finally !!!  get over it !!! 

void taskCanbus_Write(xTask task_, xTaskParm parm_) 
    {
   **char* notif_str ;** ----> change type from string to char* is working !
    xTaskNotification notif = xTaskNotifyTake(task_);

    if (notif) {
        notif_str = notif->notificationValue;
        Serial.print("Get Notify :");
        Serial.println(notif_str);
         
            Serial.println("task canbus");

            canMsg.data[0]=temperature_data.BT_AvgTemp;
            canMsg.data[1]=temperature_data.ET_CurTemp;
        
            Serial.print("CAN BUS DATA :");

            Serial.print(canMsg.data[0],HEX);
            Serial.print(" ");    
            Serial.print(canMsg.data[1],HEX);
            Serial.println(" ");
        xMemFree(notif);
            }       
    }

MannyPeterson, 2023-02-25:
@sakunamary 

Nice job! I was just going to tell you to use a char array and not an Arduino String. But you got to the answer first.

I am closing the issue for now. If you have any other issues feel free to open a new issue.

Best wishes!

closed 2023-02-25 by MannyPeterson