heliosproj/HeliOS #43: HeliOS 0.3.5 cannot compile on ESP8266

sakunamary, 2023-02-28:
Hi guys:
i am trying to compile the following code in vscode+platfromIO :
[code.zip](https://github.com/MannyPeterson/HeliOS/files/10851717/code.zip)
Not success, here is the output 

PLATFORM: Espressif 8266 (4.1.0) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.30101.0 (3.1.1) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 41 compatible libraries
Scanning dependencies...
Dependency Graph
|-- MAX6675 library @ 1.1.0
|-- HeliOS @ 0.3.5
|-- SPI @ 1.0
|-- autowp-mcp2515 @ 1.0.3
|   |-- SPI @ 1.0
Building in release mode
Compiling .pio/build/LGT8F328P_esp8266/src/main.cpp.o
Generating LD script .pio/build/LGT8F328P_esp8266/ld/local.eagle.app.v6.common.ld
Compiling .pio/build/LGT8F328P_esp8266/lib4af/MAX6675 library/max6675.cpp.o
Compiling .pio/build/LGT8F328P_esp8266/lib3ef/HeliOS/device.c.o
Compiling .pio/build/LGT8F328P_esp8266/lib3ef/HeliOS/mem.c.o
Compiling .pio/build/LGT8F328P_esp8266/lib3ef/HeliOS/port.c.o
Compiling .pio/build/LGT8F328P_esp8266/lib3ef/HeliOS/queue.c.o
Compiling .pio/build/LGT8F328P_esp8266/lib3ef/HeliOS/stream.c.o
Compiling .pio/build/LGT8F328P_esp8266/lib3ef/HeliOS/sys.c.o
Compiling .pio/build/LGT8F328P_esp8266/lib3ef/HeliOS/task.c.o
Compiling .pio/build/LGT8F328P_esp8266/lib3ef/HeliOS/timer.c.o
Compiling .pio/build/LGT8F328P_esp8266/libe14/SPI/SPI.cpp.o
Archiving .pio/build/LGT8F328P_esp8266/lib3ef/libHeliOS.a
Indexing .pio/build/LGT8F328P_esp8266/lib3ef/libHeliOS.a
Compiling .pio/build/LGT8F328P_esp8266/lib187/arduino-mcp2515-master/mcp2515.cpp.o
src/main.cpp: In function 'void setup()':
**src/main.cpp:146:37: error: invalid conversion from 'const char*' to 'const xChar*' {aka 'const unsigned char*'} [-fpermissive]
  146 |   xTask GetThermoTask = xTaskCreate("THERMO",taskThermo_Get,&BT_ArrayIndex);**
      |                                     ^~~~~~~~
      |                                     |
      |                                     const char*
In file included from src/main.cpp:4:
lib/HeliOS/src/HeliOS.h:1194:32: note:   initializing argument 1 of 'Task_t* xTaskCreate(const xChar*, void (*)(xTask, xTaskParm), xTaskParm)'
 1194 | xTask xTaskCreate(const xChar *name_, void (*callback_)(xTask task_, xTaskParm parm_), xTaskParm taskParameter_);
      |                   ~~~~~~~~~~~~~^~~~~
The same code in LGT8F328P is OK and working well ...
Please have a check ! 
Thank you very much!

MannyPeterson, 2023-03-02:
@sakunamary 

I am not sure what compiler the ESP8266 is using, but HeliOS should compile just fine. To deal with the error, can you try casting the task name as shown below?

```C
xTask GetThermoTask = xTaskCreate((unsigned const char*) "THERMO",taskThermo_Get,&BT_ArrayIndex);
```

sakunamary, 2023-03-02:
@MannyPeterson 

Here is the log output  that casting the task name in your way ,result is not working .I will try in ARDUINO IDE later .

In PlatfromIO esp8266 use 
framework-arduinoespressif8266 @ 3.30002.0 (3.0.2)
toolchain-xtensa @ 2.100300.210717 (10.3.0)

[logoutput.txt](https://github.com/MannyPeterson/HeliOS/files/10868582/logoutput.txt)

sakunamary, 2023-03-02:
Update ：
I run the same code in arduinoIDE 1.8.19. It works .

MannyPeterson, 2023-03-04:
@sakunamary 

Sorry, not sure what is going on. I test HeliOS on the ESP8266 using PlatformIO. What are you using?

sakunamary, 2023-03-06:
> @sakunamary
> 
> Sorry, not sure what is going on. I test HeliOS on the ESP8266 using PlatformIO. What are you using?

FYI:

I am using:
VSCODE 
  version: 1.75.1 (Universal)
提交: 441438abd1ac652551dbe4d408dfcec8a499b8bf
date: 2023-02-08T21:34:59.000Z
Electron: 19.1.9
Chromium: 102.0.5005.194
Node.js: 16.14.2
V8: 10.2.154.23-electron.0
OS: Darwin x64 22.3.0

PIO IDE:
ver:3.4.3
PIO core :ver6.16 

 - framework-arduinoespressif8266 @ 3.30101.0 (3.1.1) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - toolchain-xtensa @ 2.100300.210717 (10.3.0)

On a MacBook Pro with intel cpu

MannyPeterson, 2023-03-16:
@sakunamary

I will add this to the development backlog and will test when HeliOS 0.4.0 releases which is soon. Thanks for brining this to my attention!

closed 2023-05-08 by MannyPeterson