heliosproj/HeliOS #44: Need Help for an example of QueueMessage

sakunamary, 2023-03-01:
Hi guys:
   I am learning how to use Queue of HeliOS , my code is not working ,I am not sure my codes is correct.Please do me a favor to explain how to use the function Queue in HeliOS .Thanks 
 here is my code :
[main.cpp.zip](https://github.com/MannyPeterson/HeliOS/files/10857850/main.cpp.zip)

I create tasks, task1 and task2 are just sending message to queue ,and taskprint will read the message
and print it out . It's very simple .
Question 1 :So should  I suppose to  create a   xQueue Queue1 = xQueueCreate(8) ？
Question2:  How can I get the handler of Queue in Task?

void Task1(xTask task_, xTaskParm parm_) {
char userID="1";
xQueue Queue1; -------> for Queue1 handler  Am I correct ? if I omit this line ,compiler will error : Queue1 is not declared.
        if (xQueueIsQueueFull(Queue1) == false ){
                xQueueSend(Queue1,
                            1, 
                            "1" );
            Serial.println("TASK1 sended");                            
            } else  {
            Serial.println("QueueMsg is full");                    
            }            
}
     
Thanks you again!!!!
Best wishes!!!

MannyPeterson, 2023-03-04:
@sakunamary 

Good question, so there are a few ways. The most appropriate way is to create the queue in your setup() function. Then pass the queue to both tasks using the task parameter.

```C

xQueue queue1 = xCreateQueue(8);

xTask task1 = xCreateTask("TASK1", task1_main, queue1);

xTask task2 = xCreateTask("TASK2", task2_main, queue1);

```

Inside the task main functions you would cast "parm_" to an xQueue so you can call the queue system calls.

I hope that helps!

Manny

sakunamary, 2023-03-06:
@MannyPeterson  
   
   I have tried the way you show me ,but is not working .
    here is the task function:

part1:  create  xQueue in setup() and then create two task send and receive as your way:

    xQueue Queue1 = xQueueCreate(8);

    xTask task1 = xTaskCreate("TASK1", Task1,  Queue1 );
    xTask taskPrint = xTaskCreate("PRINT", TaskPrint, Queue1);

  Part 2: passthrough the      xQueue in the task function :
   void Task1(xTask task_, xTaskParm parm_) {
xQueue queue =  DEREF_TASKPARM(xQueue, parm_);   -------> passthrough  Queue1 into queue 
        if (xQueueIsQueueFull(queue) == false ){
                xQueueSend(queue,
                             1, 
                            '1' );
            Serial.println("TASK1 sended");     

            } else  {
            Serial.println("QueueMsg is full");                    
            }

  DEREF_TASKPARM(xQueue, parm_) = queue; ----------->deref queue 
  return;
}       

Task to print the queue :
void TaskPrint(xTask task_, xTaskParm parm_) {
char* str ;
byte num;
xQueue Queue1 =  DEREF_TASKPARM(xQueue, parm_);   
xQueueMessage  Msg=xQueueReceive(Queue1);

    str = Msg->messageValue;
    num = Msg->messageBytes;
    Serial.print("RECIVE Queue:");
    Serial.println(str);

    Serial.print("RECIVE byte:");
    Serial.println(num);

  DEREF_TASKPARM(xQueue, parm_) = Queue1;
  return;       
}

OUTPUT is not what I suppose  , now is 

 RECIVE Queue:  -------->empty and suppose is  1 or 2
RECIVE byte:0    --------> suppose is 0

here is the pgm after modify.
[main.cpp.zip](https://github.com/MannyPeterson/HeliOS/files/10894396/main.cpp.zip)

Thanks for your answering again!!!
Best wishes !!!

MannyPeterson, 2023-03-16:
@sakunamary 

Apologies for the delay. I will compile and try running your code.

sakunamary, 2023-03-16:
> @sakunamary
> 
> Apologies for the delay. I will compile and try running your code.

No problem at all . Don't be rush, just take your time...I am writing some examples ino file for HeliOS.

MannyPeterson, 2023-03-16:
@sakunamary thank you for your interest and contribution!

closed 2023-05-08 by MannyPeterson